KeithSloan/OpenSCAD_Alt_Import #4: OpenSCAD executable unavailable

tegger2000, 2022-04-14:
i got the error on MacOS:
20:14:47  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/veltenspagele/Library/Preferences/FreeCAD/Mod/OpenSCAD_Alt_Import/freecad/OpenSCAD_Alt_Import/importCSG.py", line 90, in open
    tmpfile=callopenscad(filename)
  File "/Users/veltenspagele/Library/Preferences/FreeCAD/Mod/OpenSCAD_Alt_Import/OpenSCADUtils.py", line 186, in callopenscad
    raise OpenSCADError('OpenSCAD executable unavailable')
<class 'OpenSCADUtils.OpenSCADError'>: 'OpenSCAD executable unavailable'

i installed OpenSCAD with brew. I didn't find a parameter for the importer, so how could i fix that ?

tegger2000, 2022-04-14:
okay, i found it. You need to replace
 <FCParamGroup Name="OpenSCAD"  />
with
<FCParamGroup Name="OpenSCAD">
            <FCText Name="openscadexecutable">/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD</FCText>
          </FCParamGroup>
in the user.cfg. It's in ~/Library/Preferences/FreeCAD/

Nick-Z0, 2023-02-22:
Did not work for me. Below are the errors it threw for each of the available import options if anyone is interested. I found it easier to remake the object I wanted.

```
22:30:01  Verbose = False
22:30:01  Processing : /Users/User/Downloads/3D models/Triomino_Parametric_Engraved_5408529/files/Triomino_engraved.scad
22:30:01  Calling OpenSCAD
22:30:01  Using Alternate OpenSCAD Importer
22:30:01  Doc Triomino_engraved useMaxFn 16
22:30:01  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 100, in open
    processCSG(doc, tmpfile)
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 175, in processCSG
    result = parser.parse(f.read())
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/ply/yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/ply/yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 1356, in p_text_action
    t = addString(t,'script',p)
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 1339, in addString
    return(t + ', ' +s+' = "'+p[3][s]+'"')
<class 'KeyError'>: 'script'
22:32:50  Verbose = False
22:32:50  Creating SCAD File Object from : /Users/User/Downloads/3D models/Triomino_Parametric_Engraved_5408529/files/Triomino_engraved.scad
22:32:50  Create SCAD File Object Triomino_engraved path /Users/User/Downloads/3D models/Triomino_Parametric_Engraved_5408529/files/Triomino_engraved.scad
22:32:50  Triomino_engraved State : ['Touched'] prop : Proxy
22:32:50  create Geometry
22:32:50  Do not process SCAD source on Document recompute
22:32:50  View Provider OnChanged : prop DisplayMode
22:32:50  View Provider OnChanged : prop DisplayMode
22:32:50  View Provider OnChanged : prop DisplayMode
22:32:50  View Provider OnChanged : prop Proxy
22:32:50  Execute Triomino_engraved keepWork False
22:32:50  shapeFrom Source File : keepWork False
22:32:50  Create Brep Triomino_engraved.scad 16
22:32:50  Active Document Triomino_engraved
22:32:50  Source : Triomino_engraved.scad
22:32:50  SourceFile : /Users/User/Downloads/3D models/Triomino_Parametric_Engraved_5408529/files/Triomino_engraved.scad
22:32:50  <Document object at 0x7fd82c3a3c00>
22:32:50  Before :  Can't open input file '/var/folders/5l/8z4lqvmn5_n_x15dq00rzt400000gn/T/Triomino_engraved.scad'!
22:32:50  
22:32:50  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importFileSCAD.py", line 65, in open
    obj.Proxy.executeFunction(obj)
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./OpenSCADObjects.py", line 273, in executeFunction
    shp = shapeFromSourceFile(obj, modules = obj.modules)
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./OpenSCADObjects.py", line 178, in shapeFromSourceFile
    brepShape = createBrep(srcObj, tmpDir, wrkSrc)
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./OpenSCADObjects.py", line 126, in createBrep
    after = e.value.rsplit(',',1)[1]
<class 'IndexError'>: list index out of range
22:33:18  Verbose = False
22:33:18  Creating Object from : /Users/User/Downloads/3D models/Triomino_Parametric_Engraved_5408529/files/Triomino_engraved.scad
22:33:18  Create Object Triomino_engraved path /Users/User/Downloads/3D models/Triomino_Parametric_Engraved_5408529/files/Triomino_engraved.scad
22:33:18  Triomino_engraved001 State : ['Touched'] prop : Proxy
22:33:18  create Geometry
22:33:18  Do not process SCAD source on Document recompute
22:33:18  Doc temp dir /var/folders/5l/8z4lqvmn5_n_x15dq00rzt400000gn/T
22:33:18  Path /var/folders/5l/8z4lqvmn5_n_x15dq00rzt400000gn/T/Triomino_engraved001.scad
22:33:18  Copy File /Users/User/Downloads/3D models/Triomino_Parametric_Engraved_5408529/files/Triomino_engraved.scad /var/folders/5l/8z4lqvmn5_n_x15dq00rzt400000gn/T/Triomino_engraved001.scad
22:33:18  Triomino_engraved001 State : ['Touched'] prop : sourceFile
22:33:18  View Provider OnChanged : prop DisplayMode
22:33:18  View Provider OnChanged : prop DisplayMode
22:33:18  View Provider OnChanged : prop DisplayMode
22:33:18  View Provider OnChanged : prop Proxy
22:33:18  Execute Triomino_engraved001 keepWork False
22:33:18  shapeFrom Source File : keepWork False
22:33:18  Create Brep Triomino_engraved001.scad 16
[... 23 lines cut ...]
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 1356, in p_text_action
    t = addString(t,'script',p)
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 1339, in addString
    return(t + ', ' +s+' = "'+p[3][s]+'"')
<class 'KeyError'>: 'script'
22:33:29  Processing : /Users/User/Downloads/3D models/Triomino_Parametric_Engraved_5408529/files/Triomino_engraved.scad
22:33:29  Calling OpenSCAD
22:33:29  Using Alternate OpenSCAD Importer
22:33:29  Doc Triomino_engraved1 useMaxFn 16
22:33:29  Traceback (most recent call last):
  File "<string>", line 1, in <module>
[... cut ...]
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/ply/yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 1356, in p_text_action
    t = addString(t,'script',p)
  File "/Users/User/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 1339, in addString
    return(t + ', ' +s+' = "'+p[3][s]+'"')
<class 'KeyError'>: 'script'
```

KeithSloan, 2023-02-22:
What happens when you try the standard importer!!! ? i.e The OpenSCAD Workbench.

The Alternate importer does not use a different method or obtaining the location of the OpenSCAD executable.

I might be wrong but my suspicion is there is a problem with the OpenSCAD Workbench settings.

KeithSloan, 2023-02-22:
Nick-Z0

Your error was not a OpenSCAD executable unavailable but a bug with Fix bug with ScadFileObject (importFileSCAD) and modules=True property'

Should be fixed in latest version.

KeithSloan, 2023-02-22:
You are also getting an error on parsing a OpenSCAD **text** function, please could you supply the SCAD source you are using or the saved FreeCAD file when you use SCADObject import.

Nick-Z0, 2023-02-22:
The file is here: https://www.thingiverse.com/thing:5408529
There is only one scad file. I could not upload this filetype here.

Nick-Z0, 2023-02-22:
> What happens when you try the standard importer!!! ? i.e The OpenSCAD Workbench.
> 
> The Alternate importer does not use a different method or obtaining the location of the OpenSCAD executable.
> 
> I might be wrong but my suspicion is there is a problem with the OpenSCAD Workbench settings.

I think it is more of a setup problem from my side. I did not find the original workbench and after I found it and tried it, there was a library missing (DXF) etc. It is my first time using this program so I did not know what to do.

KeithSloan, 2023-02-22:
Suggest you uninstall the Alternate OpenSCAD importer - Use FreeCAD | Tools | Addon Manager
Scroll to Alternate OpenSCAD workbench an **uninstall.**

Try with the standard OpenSCAD importer i.e. Just try and import the file.

You have to Activate the OpenSCAD work bench.
Click On FreeCAD | Preferences

Should see a screen like this

[image]

Click on the OpenSCAD icon in the left hand column
[image]

Should now see 

[image]

You then need to tell it where OpenSCAD is installed.

I would the restart FreeCAD just to be on the safe side.
Then try and import your file.

File | Open 

When you have the standard importer working then try and reinstall the Alternate OpenSCAD importer

FreeCAD | Tools | Addon Manager - etc.

KeithSloan, 2023-02-22:
Nick-Z0

Thanks for reporting the problem

I can recreate your problem, it also occurs on the FreeCAD OpenSCAD workbench importer, I have made some small fixes so things get further i.e. It successfully imports the DXF file that is created but there is a bug in the clean up of layers. I have PM's Chris Hennes for his thoughts, as he has been maintaining the main importer.

KeithSloan, 2023-02-23:
Have pushed updates to repro.

Triomino_engraved.scad now imports
[Triomino_engraved.FCStd.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/10809365/Triomino_engraved.FCStd.txt)

closed 2023-02-27 by KeithSloan