KeithSloan/OpenSCAD_Alt_Import #7: Question: how will OpenSCAD users benefit from this alternate importer without installing it via the addon manager?

luzpaz, 2023-02-22:
The work being done here is excellent but out of reach for OpenSCAD WB FreeCAD users. What is the short/long term vision to improve the situation ?

KeithSloan, 2023-02-22:
Installing the Alternate Importer does not stop OpenSCAD WB FreeCAD users. If it is installed and a file is opened the prompt offers the new importer and the current standard importer.

There was not much reaction/feedback on the taster I posted following the FreeCAD day, so adding function to the Alternate importer was/is a way of again testing the water.

When a OpenSCAD problem comes up, if I can fix in the Alternate Importer, then I can make the changes available very quickly with no need for merges into the Main FC OpenSCAD workbench and then users having to go to a daily version say. Again one of the first things I do with a problem is try the Alternate importer to see if there has been a regression.
Chris @chennes and I often exchange PM's so that if I do fix something he can update the main version but those updates are not going to make a main version straight away.

I don't know what Chris @chennes thinks but I think a good plan would be to add new facilities to the Main OpenSCAD workbench before the next main FreeCAD release. A very large part of the code is the same, the Alternate importer essentially just has some extra code.

One main update was to add a timeout facility for when FC calls openscad. I would like Chris to buddy check my code/update in this area. I added this facility as when I was toggling the Mode to Mesh and details of the source file get passed to OpenSCAD on the Command line for some files I was NOT getting an STL file back. Asking for an STL file back has not been used before but it is just one change to one parameter on the OpenSCAD workbench callopenscad function call. At least one file which the Native OpenSCAD application(I am on a Mac) previews very quickly and creates an STL file in not much longer, with the timeout facility,  it times out but goes on to create a FC mesh from the STL. Given this is being passed to OpenSCAD via the command line, then I am not sure if this is an OpenSCAD bug. Chris has some experience of tracking down some bugs that don't show up with the native OpenSCAD app. It would be useful to get to the bottom of this before updating the main FC OpenSCAD workbench, although if this problem occurs when OpenSCAD is called to create a CSG file which is the only option the current FC workbench exercises it might be useful before then.

The READMe for the Alternate importer asks the user to add Parameters to the OpenSCAD workbench, it would be better to have this editable in the OpenSCAD workbench parameters but I am not going there for two reasons.

1) I don't know where the Qt source is.
2) Somebody, made some changes in this area in the past and as I remember, some code and sensible options no longer worked, plus I was told the problems were my fault and I did not know what was doing.

So personally I have an aversion for making such an update. I am happy to describe the new parameters needed. 
There may also be some additional ones in the near future.

I am setting some Object properties to boolean and then actioning when they are changed to True. Would like consideration to be given to adding new property that was just action, so if clicked on it would drive the FeaturePython function OnChanged.

Also when I update the message property with any error message that comes back from calling OpenSCAD it does not update the Properties Window straight away and some times I get a warning message, so again it would be good to get to the bottom of a correct way of doing this before updating the Main FC OpenSCAD workbench.

luzpaz, 2023-02-27:
Thanks for the elaborate response. Lets see what @chennes says