KeithSloan/OpenSCAD_Alt_Import #8: Error recreating part

mconsidine, 2023-02-22:
Hi,
The error message in the text console is:
`
16:58:58  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/matt/.local/share/FreeCAD/Mod/OpenSCAD_Alt_Import/./importSCAD.py", line 67, in open
    obj.Proxy.executeFunction(obj)
  File "/home/matt/.local/share/FreeCAD/Mod/OpenSCAD_Alt_Import/./OpenSCADObjects.py", line 274, in executeFunction
    shp = shapeFromSourceFile(obj, modules = obj.modules)
  File "/home/matt/.local/share/FreeCAD/Mod/OpenSCAD_Alt_Import/./OpenSCADObjects.py", line 179, in shapeFromSourceFile
    brepShape = createBrep(srcObj, tmpDir, wrkSrc)
  File "/home/matt/.local/share/FreeCAD/Mod/OpenSCAD_Alt_Import/./OpenSCADObjects.py", line 87, in createBrep
    processCSG(wrkDoc, tmpFileName, srcObj.fnmax)
  File "/home/matt/.local/share/FreeCAD/Mod/OpenSCAD_Alt_Import/./importCSG.py", line 175, in processCSG
    result = parser.parse(f.read())
  File "/usr/lib/python3/dist-packages/ply/yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "/usr/lib/python3/dist-packages/ply/yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "/home/matt/.local/share/FreeCAD/Mod/OpenSCAD_Alt_Import/./importCSG.py", line 1056, in p_multmatrix_action
    retList.append(performMultMatrix(part,matrixisrounded,transform_matrix))
  File "/home/matt/.local/share/FreeCAD/Mod/OpenSCAD_Alt_Import/./importCSG.py", line 1110, in performMultMatrix
    new_part.Shape = part.Shape.transformGeometry(transform_matrix) 
<class 'Part.OCCError'>: gp_GTrsf::Trsf() - non-orthogonal GTrsf
`
The FC info is
[code]
OS: Linux Mint 21 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31995 (Git)
Build type: Release
Branch: master
Hash: b2ee277a94d0e07ae6288d3b4ce84111d7720aa1
Python 3.10.6, Qt 5.15.3, Coin 4.0.1, Vtk 9.1.0, OCC 7.8.0.dev
Locale: English/United States (en_US)
Installed mods: 
  * OpenSCAD_Alt_Import 1.0.0
  * ProDarkThemePreferencePack 1.0.0
  * toSketch 1.0.1
  * Render 2023.2.0
  * POV-Ray-Rendering
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * fasteners 0.4.54
  * Silk 1.0.0
  * Dracula 0.0.2
  * Curves 0.6.6
  * ThreadProfile 1.84.0
[/code]

And a minimal working problematic example is:
`
$fs = 0.01;
$fn=192;
scalef=25.4;  //25.4 for mm; 1 for imperial
Previewoffset = $preview ? 0.01 : 0;
brass = "#F7E557";

module brake_ring() //TODO: edges should be rounded
{
  color(brass) 
  {
    difference()
    {
      union()
      {
        cylinder(h=(2/8), d= (3/8));
        translate([0,0,2/8])      
          cylinder(h=(1/8), d=(3/4));
      }
      translate([0,0,0-Previewoffset])
        cylinder(h=(2/8)+(1/8)+2*Previewoffset, d=3/16);
      translate([0,0,(2/8)+(1/16)-Previewoffset])
        cylinder(h=(1/16)+2*Previewoffset, d=(3/8));
    }
  }
}

scale([scalef,scalef,scalef]) 
{
translate([4,1,0])
  brake_ring();  
}
`
This is carved out of a larger openSCAD file that has a number of parts built up from measurements on blueprints.  Thus the use of imperial units.  I can pull some of them in, but then it appears to stop with the above message.  (The use of Previewoffset is a hack to deal with a "skin" that can show up in preview mode when taking a difference ...)

The version of the importer code is as of an hour ago 16:08 NY time.

Happy to try to diagnose this further.

Regards,
mconsidine

mconsidine, 2023-02-22:
Apologies for the formatting of the above.  If there's a pointer on how to clean that up, happy to do so.
[test_for_FC.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/10808713/test_for_FC.txt)

KeithSloan, 2023-02-22:
Not seeing problem here, Looks like there might be a regression in OCC 7.8.0.dev
**(Note it says .dev)**

Please could I ask you to uninstall the Alternate Importer and try with the standard FreeCAD OpenSCAD workbench importer. If that shows a problem please raise as a Main FreeCAD issue see https://github.com/FreeCAD/FreeCAD/issues

My system info.
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ImportNURBS 1.1.0
  * ImportExport_3DM 0.1.0
  * DynamicData 2.46.0
  * Pyramids-and-Polyhedrons
  * occi-freecad-plugin 0.1.0
  * GDML 2.0.0
  * Design456 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1
  * Curves 0.6.6
[/code]
[image]

[Issue8.scad.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/10808843/Issue8.scad.txt)
[issue8.FCStd.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/10808844/issue8.FCStd.txt)

mconsidine, 2023-02-22:
Thanks for the review.
Looks like I'm getting the same thing using the standard importer.  So my next step should be to downgrade OCC and see if anything changes.
Thanks again - I'm closing this as "not your problem" ...
mconsidine

KeithSloan, 2023-02-23:
Glad you got things sorted.

I assume you imported as a SCADFileObject or a SCADObject ?

If you just selected SCAD (ImportCSG) then this should cause them to be converted into FreeCAD Brep shapes which you should be able to access individually but they will no longer be tied to their OpenSCAD scad source.

Could you point me to the source file and I will look at if there is anything I can do.

mconsidine, 2023-02-23:
[EquatorialClockDrive_vFC.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/10819140/EquatorialClockDrive_vFC.txt)

Thanks.  I'm attaching here the file I'm working on, which is a number of pieces derived from blueprints.  So it is sort of my stress-test for going from openSCAD to FC.

Currently presenting a challenge is the use of the thread library.  I think - but am not certain - that defining threads is a bit more involved in FC than openSCAD.  If I uncomment the calls to "english_thread" I get a) long wait time and b) a mis-sized thread of some sort.

The angled brackets come in nice initially as objects, but if I "explode compound"  I get artifacts.

A minor nit is that somehow two pieces colored in red don't come in that way, though it appears colors were correctly applied early on.

Finally, two "edge" uses are putting extruded letters on a curve and pulling a shape in from a .DXF that then gets extruded.

I'm really impressed with this ability to pull in openSCAD work that I've done in the past.  And I get that there's some stuff that perhaps should just be done in FC.  So I hope nothing in the above is taken as a complaint/criticism.

If you've got any thoughts on the attached (other than noting my coding style probably leaves much to be desired), I'd be appreciative.  If not, thanks anyway!

mconsidine

KeithSloan, 2023-02-23:
Thanks - my plan was to start to look at Libraries to see what I can do but I think it might mean i have to code a lex scad parser which will take some time. Todate, what the importer is uses is a lex csg parser.

I will close this issue now and get back in touch if I can make any progress.

KeithSloan, 2023-02-23:
Please could you advise where I find the threads library you use.

I have installed

- https://github.com/rcolyer/threads-scad
-https://github.com/adrianschlatter/threadlib

mconsidine, 2023-02-23:
I believe this is what I used  https://dkprojects.net/openscad-threads/If there were something as straightforward but more friendly for converting to FC, I'd be happy to switch.  mconsidine

https://github.com/rcolyer/threads-scad
-https://github.com/adrianschlatter/threadlib

closed 2023-02-24 by KeithSloan

mconsidine, 2023-02-24:
> Thanks - my plan was to start to look at Libraries to see what I can do but I think it might mean i have to code a lex scad parser which will take some time. Todate, what the importer is uses is a lex csg parser.
> 
> I will close this issue now and get back in touch if I can make any progress.

You've probably seen this
https://github.com/jeff-dh/py_scadparser
but I pass along regardless.  Have not tested it out - perhaps today.  Thank you for your efforts.

KeithSloan, 2023-02-24:
Thanks - Had not seen and think it will be useful.

It is as per,  I used for csg based on the old way of doing things and I would like to start using the python lex library in the more Object Oriented manner.