KeithSloan/OpenSCAD_Alt_Import #10: Error importing a DXF file

mconsidine, 2023-03-17:
In FC, I can select a DXF file via File -> Open and it will pull the image in.

If I load the OpenSCAD WB and select Add OpenSCAD Element and enter
  import("./testbend.dxf");
then I get the following error:

`15:17:10  Using Alternate OpenSCAD Importer
15:17:10  Doc testbend useMaxFn 16
15:17:10  Traceback (most recent call last):
15:17:10    File "/home/matt/.local/share/FreeCAD/Mod/Alternate_OpenSCAD/./OpenSCADCommands.py", line 404, in addelement
15:17:10      importCSG.insert(tmpfilename,doc.Name)
15:17:11    File "/home/matt/.local/share/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 137, in insert
15:17:11      processCSG(doc, filename)
15:17:11    File "/home/matt/.local/share/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 176, in processCSG
15:17:11      result = parser.parse(f.read())
15:17:11    File "/usr/lib/python3/dist-packages/ply/yacc.py", line 333, in parse
15:17:11      return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
15:17:11    File "/usr/lib/python3/dist-packages/ply/yacc.py", line 1120, in parseopt_notrack
15:17:11      p.callable(pslice)
15:17:11    File "/home/matt/.local/share/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 900, in p_import_file1
15:17:11      p[0] = [process_import_file(filen,ext,p[3]['layer'])]
15:17:11  KeyError: 'layer' `

The section of code it is referring to in importCSG.py is:
`def p_import_file1(p):
    'import_file1 : import LPAREN keywordargument_list RPAREN SEMICOL'
    if printverbose: print("Import File")
    filen,ext =p[3]['file'].rsplit('.',1)
    p[0] = [process_import_file(filen,ext,p[3]['layer'])]
    if printverbose: print("End Import File")`

and I'm guessing that it is expecting a layer to be named "layer".

The test file was created with Inkscape and one version (using "R12" DXF format) doesn't allow for any metadata to be set when saved.  Using the "R14" version does.  So testbend2.dxf is saved with that format and I (thought I) set the layer name to "layer".

In neither case can I import it with the "import" command, but in both cases I can pull it into a workspace via File -> Open

Is this expected behavior or is there something I should be doing differently?

TIA,
mconsidine
[testbendDXF.zip](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/11005230/testbendDXF.zip)

mconsidine, 2023-03-17:
Sorry about the formatting being messed up.  If there's a preferred way to have messages inserted, please advise!
mconsidine

KeithSloan, 2023-03-17:
Have asked Chris Hennes to look at this as the problem occurs the same in the standard importer.
https://forum.freecad.org/viewtopic.php?t=76892

mconsidine, 2023-03-17:
Thanks for checking on this.
mconsidine

Sent from allegedly smart phone

KeithSloan, 2023-03-18:
See the latest update to the post in the Forum.

What I am seeing now is

(Note I changed Layer 1 to Layer1 to eliminate the space as a possible error)
````
1:05:16  Processing : /Users/keithsloan/Downloads/testImport.csg
11:05:16  Using Alternate OpenSCAD Importer
11:05:16  Doc testImport useMaxFn 16
11:05:16  Import File {'file': 'testBend.dxf', 'layer': 'Layer1', 'origin': [0.0, 0.0], 'scale': '1', 'convexity': '1', '$fn': '0', '$fa': '12', '$fs': '2', 'timestamp': '1679136770'}
11:05:16  Process DXF file
11:05:16  File Name : testBend
11:05:16  Layer : Layer1
11:05:16  PathName : /Users/keithsloan/Downloads
11:05:16  DXF Full path : /Users/keithsloan/Downloads/testBend.dxf
11:05:16  opening /Users/keithsloan/Downloads/testBend.dxf...
11:05:16  ----------------
11:05:16  Layer
11:05:16  name: 0
11:05:16  line_color: (0.0, 0.0, 0.0)
11:05:16  line_width: 2.0
11:05:16  draw_style: Solid
11:05:16  transparency: 0
11:05:16  drawing 106 lines...
11:05:16  ----------------
11:05:16  Layer
11:05:16  name: Layer1
11:05:16  line_width: 2.0
11:05:16  draw_style: None
11:05:16  typecheck[make_layer]: 'None' is not <class 'str'>
11:05:16  Wrong input: must be 'Solid', 'Dashed', 'Dotted', or 'Dashdot'.
11:05:16  Traceback (most recent call last):
````
So it is not finding a line type for a Layer1 item.

I think this maybe a bug in FreeCAD's Draft importDXF that it should be setting a default.

Does Inkscape have away of setting a default draw_style or what if you create your Inkscape file with a draw_style?

KeithSloan, 2023-03-18:
Oh. I forgot, I created a test scad file as
````
import("/Users/keithsloan/Downloads/testBend.dxf");
````
Easier for testing.
I used OpenSCAD to create a csg file which produced
````
import(file = "testBend.dxf", layer = "", origin = [0, 0], scale = 1, convexity = 1, $fn = 0, $fa = 12, $fs = 2, timestamp = 1679136770);
````
And edited it to look like
````
import(file = "testBend.dxf", layer = "Layer1", origin = [0, 0], scale = 1, convexity = 1, $fn = 0, $fa = 12, $fs = 2, timestamp = 1679136770);
````

But your error indicates that the importer was not finding a 'layer' specification in the csg file, so maybe that is due to
the fact that you used AddElement.

Please could you also try with a scad file. You should even be able to create it by the 'Save' button in AddElement

mconsidine, 2023-03-18:
I exported it from openSCAD as a CSG file and looked at the contents.  Mine didn't have a "layer" field.  But when I manually added it I was able to reproduce your error from above.

I think I'll step back to the original JPG that I worked from and document how that was turned into a DXF via Inkscape.  Maybe there's a better tool or workflow to use to create a FC-friendly DXF?  I'm certainly open to that.   But I was surprised since I could File -> Import or Open the DXF in FC without having to enable the Draft WB that I ran into issues.

I'll try to recreate the DXF and better pay attention to what is/isn't a setting in Inkscape.
mconsidine

mconsidine, 2023-03-18:
Ok, this works in FC via Add OpenSCAD Element within the OpenSCAD WB:
  import("/home/matt/Downloads/3D_design_files/testbend_plainsvg.svg", layer="");

Omitting 'layer=""' generates an error (where its looking for a keyword).  So too does using DXF file formats for input.  Even with 'layer=""', the DXF file seems to generate an error.

Oddly, adding 'layer=""' in OpenSCAD doesnt generate an error, but doesn't load the shape.  (I'm using a recent build of OpenSCAD, fwiw).

So I'm confused.  The OpenSCAD page here
  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Importing_Geometry#import
says that "layer" is optional.

The challenge is that the DXF files were coming from Inkscape, not OpenSCAD.  It seems, however, if I generate SVG files and import those, I'm okay.  As long as I use a paramenter that OpenSCAD considers optional. 

If I have muddied things by overlooking some recommended workflow I'd be happy to adopt that.  But it seems like there's some inconsistency somewhere and I don't know where it lies to help fix it.

KeithSloan, 2023-03-18:
If it works in OpenSCAD we should also try and make it to work.

The way the importer works is to create a csg file under the covers, it then parses the csg file with python version of lex and acts on the commands on the csg creating the FC equivalent shapes.(Apart for hull and minkowski)
Now OpenSCAD has always added the layer option on creating the csg and the importer has this hard coded.

What version of OpenSCAD are you running? I am on OpenSCAD 2021.01. ( On a Mac )

Here is a test scad file that i created  ( You need to drop the txt extension but we cannot upload scad file to github )
[testImport.scad.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/11009763/testImport.scad.txt)

Here is the CSG file that OpenSCAD produces
[testImport.csg.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/11009764/testImport.csg.txt)
i.e. with layer=""

Can you check what the file produces on your system, if you load into OpenSCAD and export as CSG.

I know we need to get this to work with AddElement but it stores the under the cover csg file deep inside FC,
so best to test with a file, until we can find the problem.

KeithSloan, 2023-03-18:
When you say "Even with 'layer=""', the DXF file seems to generate an error."

With your test file I am seeing an error in Report View
````
11:05:16  Layer
11:05:16  name: Layer1
11:05:16  line_width: 2.0
11:05:16  draw_style: None
11:05:16  typecheck[make_layer]: 'None' is not <class 'str'>
11:05:16  Wrong input: must be 'Solid', 'Dashed', 'Dotted', or 'Dashdot'.
11:05:16  Traceback (most recent call last):
````
So there is probably a bug with the FC DXF importer as it should really use a default if 'draw_style' is not set.
We need to pursue this in the FreeCAD forum see [https://forum.freecad.org/viewtopic.php?t=76892&sid=1e7e8367a07fc7f53e14e854fabc1a36#p668494](url)

chennes, 2023-03-24:
I've submitted a PR that includes code to handle any of those parameters being NULL: https://github.com/FreeCAD/FreeCAD/pull/9021

With this I can successfully import the test DXF file, though the OpenSCAD processing fails later on.

KeithSloan, 2023-03-24:
Okay I have looked at the code and it is trying to fuse an empty list of faces.

If I backoff a couple of functions to line 410 edgestofaces

Then the code is 
````
def edgestofaces(edges,algo=3,eps=0.001):
    #edges=[]
    #for shapeobj in (objs):
    #    edges.extend(shapeobj.Shape.Edges)
    #taken from Drafttools
    #from draftlibs import fcvec, fcgeo
    import Part
    #wires = fcgeo.findWires(edges)
    wires = edgestowires(edges,eps)

Must be missing something as I cannot find Drafttools, no idea what the various algorithims is about.

And at the bottom of the function there is

````
    if algo is None:
        return facel
    elif algo == 1: #stable behavior
        return subtractfaces(facel)
    elif algo == 0: #return all faces
        return Part.Compound(facel)
    elif algo == 2:
        return subtractfaces2(facel)
    elif algo == 3:
        return Overlappingfaces(facel).makeshape()
````

If I deduce correctly it is trying to use algo = 3

Tried algo = 1 still fails

If I Back off some more to line 405 and print out edges it is an empty list, so there is something about the DXF processing that to producing an empt list of edges.

WE REALLY do need a goox DXF import Library that supports import to a Face

mconsidine, 2023-03-24:
If I read that code right, it looks like it used to pull in "draftlibs" from Draft, which seems to be where fcgeo lives.  But then similar functionality from Part was substituted. 

I don't know the history of all that, but maybe it could be reverted?  I will try that ...
mconsidine

KeithSloan, 2023-03-24:
"I don't know the history of all that, but maybe it could be reverted? I will try that ..."

OpenSCAD2Dgeom.py will be Sebastian Hoogen's code dating possibly back to 2013.

For the OpenSCAD importer when it encounters a importDXF function, it needs to import the DXF file into a Face,
i.e. Object's Shape ( In this case a Face) so that it can be put on the parsers object stack, for further processing/parsing.

It looks to me like the code is an adaption of the Draft code. But I think at the end of the day, like the lack of draw_style default, there is something that is not being handled correctly and is producing an empty list of edges.

mconsidine, 2023-03-26:
Up front, I'll concede that what is below is both a hack and (probably) heresy.  That said, it is working for me.
1)  My understanding from the openSCAD docs was that the "layer" parameter to the import command was optional.  So in my scripts I had not used it.  It looks like the code in FC assumes that it is specified.  So I tweaked the relevant routine to try to accomodate the case when it doesn't exist.  This goes in importCSG.py:
`def p_import_file1(p):
    'import_file1 : import LPAREN keywordargument_list RPAREN SEMICOL'
    if printverbose: print("Import File")
    print("HERE1")
    #MattC p[0]=None; p[1]='import'; p[2]='(', p[3]=args, p[4]=')', p[5]=';'
    filen,ext =p[3]['file'].rsplit('.',1) #MattC would fail if '.' is elsewhere in path/name
    if 'layer' in p[3]: #MattC make sure 'layer' as a keyword is present in dictionary
        if len(p[3]['layer']) == 0:  #MattC may be overkill but what the hell
          layername=None
        else:
          layername=p[3]['layer']
    else: #MattC 'layer' is supposedly an optional param in straight OpenSCAD code so trap it
        layername=None
    p[0] = [process_import_file(filen,ext,layername)]  #MattC now we know we get here legit...
    if printverbose: print("End Import File")
 `
2) Parenthetically, while trying to find a typo I made I noticed that line 1294 of importCSG.py has a ";" in it.  Is that supposed to be there?  Not saying that it shouldn't, but leaving one in as a leftover from working in OpenSCAD caused me a headache ...  fwiw.

3)  And now for the heresy/idea.  Generally, the DXF files I'll deal with are related to scans of old blueprints that I then want to trace or get into some vector format.  In any case, I finally got "import" to work on my test cases (with one exception) by replace the  "importDXFface" code in OpenSCAD2Dgeom.py with this

`def importDXFface(filename,layer=None,doc=None):
    import ezdxf
    import Part
    fdoc = doc or FreeCAD.activeDocument()
    ddoc = ezdxf.readfile(filename)
    msp = ddoc.modelspace()    
    lines = msp.query('LINE')
    splines = msp.query('SPLINE')
    plines = msp.query('POLYLINE')
    lwplines = msp.query('LWPOLYLINE')
    #Not sure if order of elements in file matters??
    faces=list()
    face_l=None
    face_p=None
    face_lwp=None
    
    if len(lines)>0:
        elements = []
        for l in lines:
            if l.dxf.start != l.dxf.end:
                poly = Part.makeLine(tuple(l.dxf.start), tuple(l.dxf.end)) #makes Edge object
            else:
                print("START==END")
            #poly = Part.Wire(poly) #makes Wire object
            elements.append(poly)  #creates list of Wires for use in Part.Face
        face_l=Part.Wire(elements) #L
        
    '''
    if len(splines)>0:                  
        for s in splines:
            print("SPLINE")
            #s.control_points
            #s.fit_points
    '''
    
    if len(plines)>0:
        for p in plines:
            #print("POLYLINE")
            verts = p.vertices
            verts = [tuple(v.dxf.location) for v in verts]
            if p.is_closed:
                verts += [verts[0]]
            poly = Part.makePolygon(verts)
        face_p=poly #P
    
    if len(lwplines)>0:
        elements=[]
        for lwp in lwplines:
            #print("LWPOLYLINE")
            with lwp.points("xy") as points: #xyseb
                for p in points: #TODO: vectorize??
                    poly = tuple([p[0], p[1], 0])
                    elements.append(poly)
                poly = Part.makePolygon(elements) #makes a Wire
        face_lwp=poly #LWP
        
    if face_l != None:
        faces.append(face_l)
    if face_p != None:
        faces.append(face_p)
    if face_lwp != None:
        faces.append(face_lwp)
        
    faces=Part.Face(faces)
    
    #if (len(ddoc.entities) != len(lines)+len(splines)+len(plines)+len(swplines):
    #    print("NOT ALL ENTITIES DEALT WITH")  

    return faces `

Obviously it only deals with a limited number of entities but it shows a framework that could probably be filled out.  And it imports the "ezdxf" library to do the lifting.  I have a sense that there was some reason why that library hasn't been leveraged in the FC distribution, but it's above my paygrade to argue about that.  If I can get the "SPLINE" type to work then I have a feeling 99% of my use cases will be covered.

Anyway, I pass this along in the event that its helpful.  When/if I get to a point of being incredibly proficient in using FC, I may not need this bridge.

mconsidine

KeithSloan, 2023-03-28:
Hi Matt?

Have started a discussion with yorik and chennes about use of ezdxf library.
Yorik's initial response

"Yes, definitely ezdxf is the best on the field now. At the time I started working on DXF import it wasn't there.

You would basically mean coding a new FreeCAD DXF importer from scratch? That's not a bad idea... The old python DXF importer is really a PITA and the new C++ one is fast but not very easily customizable."

A new importer based on ezdxf is not a fix in the short term.

Have tried to create a new branch **mconsidine** with my attempt to accommodate your changes, with the following changes

1) If layer is not coded in the source the csg will be created with layer=''
2) Rather than change OpenSCAD2Dgeom.py I have put the importDXFface function in a new file OpenSCADdxf.py
(OpenSCAD2Dgeom.py is part of standard FC and I don't want to make changes to that at this stage)
3) I have added the requirement of ezdxf to the metadata, to get it installed into a python library that FC has access to.

But I have not got something correct as when i try and import testimport.scad I get the following

````
12:03:40  ('Mod/Pyramids_and_polyhedrons', 'Mod/Pyramids-and-Polyhedrons', 'Mod/FreeCAD-Pyramids-and-Polyhedrons', 'Mod/Pyramids-and-polyhedrons')
12:04:37  Verbose = False
12:04:37  Processing : /Users/keithsloan/Downloads/testImport.scad
12:04:37  Calling OpenSCAD
12:04:37  Using Alternate OpenSCAD Importer
12:04:37  Doc testImport useMaxFn 16
12:04:37  import EZDXFface layer = None
12:04:37  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 101, in open
    processCSG(doc, tmpfile)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 176, in processCSG
    result = parser.parse(f.read())
  File "/Applications/FreeCAD_0.20.2.app/Contents/Resources/lib/python3.10/site-packages/ply/yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "/Applications/FreeCAD_0.20.2.app/Contents/Resources/lib/python3.10/site-packages/ply/yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 908, in p_import_file1
    p[0] = [process_import_file(filen, ext, layerName)]
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 928, in process_import_file
    obj=processDXF(fname, layer)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 1028, in processDXF
    face = importEZDXFface(filename,layer,doc)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./OpenSCADdxf.py", line 27, in importEZDXFface
    face_l=Part.Wire(elements) #L
<class 'Part.OCCError'>: BRep_API: command not done

````

Are you able to try out the branch mconsidine?

mconsidine, 2023-03-28:
Ps can you provide your testscad.scad?
Thx,
mconsidine

mconsidine, 2023-03-28:
Reviewing quickly your error msgs, I'm not sure at the moment why that failed.  Will review/test ...

KeithSloan, 2023-03-28:
[testbend.dxf.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/11090206/testbend.dxf.txt)
[testImport.scad.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/11090208/testImport.scad.txt)

mconsidine, 2023-03-28:
Thanks for those.  This looks like it's related to a challenge I have with an inkscape generated dxf still.  I think there are layer names being created in newer file formats, but I didn't account for that in the parsing.  So what I need to do is also poll the file for a list of layers and then parse entities on a layer by layer basis.  I think.  In any case, I confirm your exception and will work on it.  It was on the to do list anyway ...
mconsidine

mconsidine, 2023-03-29:
Can you tell me how you created/saved testbend.dxf?  It looks like it has a change in layer names after the first 5 points, but somehow it thinks there's only 1 layer?

mconsidine, 2023-03-29:
It appears that there are 5 stray lines visible above the main shape, if you open it in LibreCAD and zoom out.  So, in a sense, there are two unconnected entities, seemingly on different layers.  If I delete that set of lines (and as a result, that layer) then I can read the file.
What I'm puzzled by is why I couldn't see it in layer-aware code that came from the docs.  I would think I should be able to go layer-by-layer and pick up these unconnected entities.  So ... more work to do.
mconsidine

KeithSloan, 2023-03-30:
> Can you tell me how you created/saved testbend.dxf? It looks like it has a change in layer names after the first 5 points, but somehow it thinks there's only 1 layer?

Thought the file testbend.dxf came from you @mconsidine

mconsidine, 2023-03-30:
Ok, I'll try to figure out where that came from.  At the moment I'm trying to make sure the layer recognition works and then I need to figure out how to code a spline that can be a wire.mconsidine

mconsidine, 2023-04-01:
Hi,
I've taken a stab at pulling elements from a DXF into FC via a variant of the above heresy.  This can be tripped up on badly formed DXF files, but I post it here in the event it's helpful.  It looks like one of my "testbend" files had somehow gotten corrupted, in that it had unconnected lines on one layer and the shape on another.  If it is fed to this new routine with the proper layername specified, it seems to work.  Would appreciate hearing any feedback.  This covers LINE, SPLINE, POLYLINE and LWPOLYLINE DXF elements.
mconsidine
[testroutine.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/11129909/testroutine.txt)

KeithSloan, 2023-04-01:
Hi Matt,

Have tried to update the mconsidine branch with your code.

Have also included the test files in a directory testcases.

Not sure if it is because I have a corrupted textbend.dxf that you talked about  but when I try with the new code I still get an error.
````
17:30:10  ('Mod/Pyramids_and_polyhedrons', 'Mod/Pyramids-and-Polyhedrons', 'Mod/FreeCAD-Pyramids-and-Polyhedrons', 'Mod/Pyramids-and-polyhedrons')
17:30:27  Verbose = False
17:30:27  Processing : /Users/keithsloan/Downloads/mconsidine/testImport.scad
17:30:27  Calling OpenSCAD
17:30:27  Using Alternate OpenSCAD Importer
17:30:27  Doc testImport useMaxFn 16
17:30:27  ALL LAYER NAMES :  ['Layer1' 'main1']
17:30:27  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 101, in open
    processCSG(doc, tmpfile)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 176, in processCSG
    result = parser.parse(f.read())
  File "/Applications/FreeCAD_0.20.2.app/Contents/Resources/lib/python3.10/site-packages/ply/yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "/Applications/FreeCAD_0.20.2.app/Contents/Resources/lib/python3.10/site-packages/ply/yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 908, in p_import_file1
    p[0] = [process_import_file(filen, ext, layerName)]
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 928, in process_import_file
    obj=processDXF(fname, layer)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 1028, in processDXF
    face = importEZDXFface(filename,layer,doc)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./OpenSCADdxf.py", line 113, in importEZDXFface
    faces = Part.Wire(elements)
<class 'Part.OCCError'>: BRep_API: command not done

mconsidine, 2023-04-01:
That's a corrupted DXF.  If you try my new routine and specify "main1" as the layer, I believe it will work.  Attached contains 2 versions that should work.

Also, I didn't see that you had done this before taking a first stab (ever, actually) at creating a pull request.

Can you test with the attached?

[testbend_files.zip](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/11130085/testbend_files.zip)

KeithSloan, 2023-04-01:
Your pull request had merge conflicts, I have tried to correctly merge but may have made errors

Now get error
````
18:16:21  ('Mod/Pyramids_and_polyhedrons', 'Mod/Pyramids-and-Polyhedrons', 'Mod/FreeCAD-Pyramids-and-Polyhedrons', 'Mod/Pyramids-and-polyhedrons')
18:16:38  Verbose = False
18:16:38  Processing : /Users/keithsloan/Downloads/mconsidine/testImport.scad
18:16:38  Calling OpenSCAD
18:16:38  Using Alternate OpenSCAD Importer
18:16:38  Doc testImport useMaxFn 16
18:16:38  ALL LAYER NAMES :  ['Image']
18:16:38  face None
18:16:38  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 101, in open
    processCSG(doc, tmpfile)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 176, in processCSG
    result = parser.parse(f.read())
  File "/Applications/FreeCAD_0.20.2.app/Contents/Resources/lib/python3.10/site-packages/ply/yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "/Applications/FreeCAD_0.20.2.app/Contents/Resources/lib/python3.10/site-packages/ply/yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 908, in p_import_file1
    p[0] = [process_import_file(filen, ext, layerName)]
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 928, in process_import_file
    obj=processDXF(fname, layer)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 1031, in processDXF
    obj.Shape=face
<class 'TypeError'>: {'sclassname': 'N4Base9TypeErrorE', 'sErrMsg': "type must be 'Shape', not NoneType", 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': "type must be 'Shape', not NoneType", 'btranslatable': False, 'breported': False}

KeithSloan, 2023-04-01:
With testbend_inkscape_nolwp.dxf

Get
````
18:24:29  Source : testImport.scad
18:24:29  SourceFile : /Users/keithsloan/Downloads/mconsidine/testImport.scad
18:24:29  <Document object at 0x7fe3c7ba4c00>
18:24:29  CSG File name /var/folders/cj/z259fmwd41dgzl8mq8nppwd40000gn/T/testImport.csg
18:24:29  Using Alternate OpenSCAD Importer
18:24:29  Doc work2 useMaxFn 16
18:24:29  ALL LAYER NAMES :  ['Image']
18:24:29  START==END?
18:24:29  CURRENT START/END :  (1.731612, 0.006809, 0.0) (1.731612, 0.006809, 0.0)
18:24:29  PRIOR START :  2.122501 0.012414
18:24:29  PRIOR END :  1.731612 0.006809
18:24:29  FIRST START :  1.731612 0.006809
18:24:29  FIRST END :  1.305901 0.102836
18:24:29  face None
18:24:29  Traceback (most recent call last):
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./OpenSCADObjects.py", line 246, in onChanged
    self.executeFunction(fp)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./OpenSCADObjects.py", line 275, in executeFunction
    shp = shapeFromSourceFile(obj, modules = obj.modules)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./OpenSCADObjects.py", line 180, in shapeFromSourceFile
    brepShape = createBrep(srcObj, tmpDir, wrkSrc)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./OpenSCADObjects.py", line 88, in createBrep
    processCSG(wrkDoc, tmpFileName, srcObj.fnmax)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 176, in processCSG
    result = parser.parse(f.read())
  File "/Applications/FreeCAD_0.20.2.app/Contents/Resources/lib/python3.10/site-packages/ply/yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "/Applications/FreeCAD_0.20.2.app/Contents/Resources/lib/python3.10/site-packages/ply/yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 908, in p_import_file1
    p[0] = [process_import_file(filen, ext, layerName)]
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 928, in process_import_file
    obj=processDXF(fname, layer)
  File "/Users/keithsloan/Library/Application Support/FreeCAD/Mod/Alternate_OpenSCAD/./importCSG.py", line 1031, in processDXF
    obj.Shape=face
<class 'TypeError'>: {'sclassname': 'N4Base9TypeErrorE', 'sErrMsg': "type must be 'Shape', not NoneType", 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': "type must be 'Shape', not NoneType", 'btranslatable': False, 'breported': False}````

mconsidine, 2023-04-01:
Ok.  Will pull your version and check.  Thx.

mconsidine, 2023-04-01:
I'm not seeing
    return faces
at the end of OpenSCADdxf.py

Am I missing something?

mconsidine, 2023-04-01:
When I add "return faces" to the install of the latest here, I get testbend_R12.dxf to load.  The other file
  testbend_inkscape_nolwp.dxf
loads, but there's something goofy about the way it came out of inkscape.  There's a "line" that has the same start and end point, and Part.makeLine doesn't like that.  So I skip over it, but then it seems to render with the fill being on the outside of the lines, rather than the inside.  I'd be inclined to add the "line" as a Point, but I'm pretty sure that won't get converted to a Wire successfully.  So that one needs some thinking ...

KeithSloan, 2023-04-01:
Not sure how the return faces got dropped it was there in the pull request, so my Bad when I did the merge conflicts.
Was just about to push an update when I found a bug ( Nothing to do with dxf ). Will try and fix it then push an update.

KeithSloan, 2023-04-02:
I started a thread in the FreeCAD forum see [https://forum.freecad.org/viewtopic.php?t=77294](url)
and already @Roy_043 posted some info on problem dxf files.

mconsidine, 2023-05-10:
Closing this thread as we're dealing with new code base.

closed 2023-05-10 by mconsidine