KeithSloan/OpenSCAD_Alt_Import #12: Clean up handling of imports ( DXF, SVG etc ) for OpenSCAD objects

KeithSloan, 2023-04-01:
Clean up handling of imports ( DXF, SVG etc ) for OpenSCAD objects.
Refactor code.

mconsidine, 2023-04-05:
Fwiw, I'm making progress.  I can import into FC a DXF that OpenSCAD can't handle.  And I'm using the DXFs that are examples in the "tests" area of its source as hurdle cases.  If there's a 2D DXF that someone has wanted to pull into FreeCAD or OpenSCAD - and can point me to it - that would be helpful.

KeithSloan, 2023-04-07:
If you can let me have your latest version, I can update the Alternate Importer and update the master branch or better still a pull request, hopefully this time without merge issues.

mconsidine, 2023-04-08:
You might have to hold your nose when you look at this, but here's where I stand.  See attached, a .py file saved with a .txt extenstion.  I'll try to put together a pull request.  Thanks for looking.
mconsidine
[OpenSCADdxf.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/11184307/OpenSCADdxf.txt)

mconsidine, 2023-04-08:
I think what I've submitted is "drop in" and conforms to current defaults.
But if FC can handle being returned a list and/or a compound then I think I
have this set up to deliver that.  The idea was to deliver all the entities
on a specified alter, or barring that, all entities on all the layers.

I tested it with a bunch of the dxf files used as tests for openscad.
There are a few that are still problematic, but I'm not sure even openscad
handles them well.  I don't think openscad can handle splines (thought I
read that somewhere), but this does.

Hope this helps.
Matt

Sent from allegedly smart phone

mconsidine, 2023-04-10:
I think I've cleaned my effort up a bit.  At least this time ellipses generally seem to work.  See attached.

The way I have tested this is to pull the attached into the FC python console and then run this
<filepath="/path/to/openscad-source/tests/data/dxf/"
filenames=["arc.dxf",
    "circle.dxf",
    "circle-advanced.dxf",
    "circle-double.dxf",
    "circle-small.dxf",
    "ellipse.dxf",
    "ellipse-arc.dxf",
    "ellipse-reverse.dxf",
    "ellipse-rot.dxf",
    "lwpolyline.dxf",
    "lwpolyline2.dxf",
    "lwpolyline-closed.dxf",
    "null-polygons.dxf",
    "open-polyline.dxf",
    "polygon8.dxf",
    "polygon-concave.dxf",
    "polygon-concave-hole.dxf",
    "polygon-concave-simple.dxf",
    "polygon-holes-touch.dxf",
    "polygon-intersect.dxf",
    "polygon-many-holes.dxf",
    "polygon-riser.dxf",
    "polygons.dxf"]

for fn in filenames:
  Part.show(importEZDXFface(filepath+fn))
>
Before doing that I uncomment lines near the end (lines 329-336) that show all the faces that have been created.  As presented, it only returns the first of the ones made, in keeping with the idea that the current mode seems to expect one dxf  with only one shape in it.
This is probably the last I'll do on this for awhile, unless I hear of an issue, suggestion, etc.  (I'd note that I didn't try to deal with TEXT-type entities, blocks, linewidths, etc etc)
[OpenSCADdxf.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/11192174/OpenSCADdxf.txt)

KeithSloan, 2023-04-10:
Thanks - I hope to have a look at objectifying your code i.e. use of Python class etc, think with your functions, should be relatively straight forward and having test dxf is very good. I assume they are in your repro? So you taking a break will be good.

mconsidine, 2023-04-10:
I don't have those test dxfs in my repo but will put them up there.  There
are a handle that don't seem to work, either because they make use of some
tag or function or ... my logic doesn't catch everything.  E.g. there's one
where there are two polygons overlapping each other (point for point) on
the same layer as I recall.  I didn't try to catch that.  Ditto for the
"MESH" tag.

And other than the malformed versions of the original dxf that I had, all
those work.

Where I tried to put in effort that isn't necessarily able to be used yet
is the idea of returning to FC a list of faces by layer.  So maybe there's
some advantage to that; maybe not.  The thought occured also that
specifying a list of layers to exclude (rather than include) might be
useful to someone.  But I didn't try to add any logic for that.

I'll get those test dxfs uploaded in a little while.
mconsidine

mconsidine, 2023-04-10:
Ok, I've zipped up all the test dxfs included in the openscad sourcecode.  That zip file can be found in the 'Notes' folder of my copy of the mconsidine branch.  Also adding them here.
Note that this collection also includes ones that failed on me...
mconsidine
[openscad_test_dxfs.zip](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/11193730/openscad_test_dxfs.zip)

mconsidine, 2023-04-11:
I've collected a bunch of dxf files that have been posted to the forum over
the years.  Looks like I let some typos get into the routine for polyline,
which hadn't been really stress tested anyway.

So I'll work on cleaning that up and will posted a revision in a few days.
Fwiw, the "polygon" routine should called for "polyline".  Somehow I goofed
in transcribing ...

Matt

Sent from allegedly smart phone

mconsidine, 2023-04-12:
I've made further updates to the code, which should address problems with the ELLIPSE and POLYLINE types.  That code is attached here and the prior versions in comments above have been deleted.
[OpenSCADdxf.txt](https://github.com/KeithSloan/OpenSCAD_Alt_Import/files/11213305/OpenSCADdxf.txt)

Note that this version of the code returns a list of faces and also a compound object.

Also, I've uploaded the test cases I found on the forum to the "Notes" section of my version of the repository.  Not included in the ZIP but uploaded anyway are two large DXFs related to a Nintendo board/box layout.  Which, to my surprise,  worked.  Others, however ... not so much.  My test notes for both the openscad test files and the forum dxfs follow:

`filepath1="/path/to/openscad-source/tests/data/dxf/"
filenames1=["arc.dxf",
    "circle.dxf",
    "circle-advanced.dxf",
    "circle-double.dxf",
    "circle-small.dxf",
    "ellipse.dxf",
    "ellipse-arc.dxf",
    "ellipse-reverse.dxf",
    "ellipse-rot.dxf",
    "lwpolyline.dxf",
    "lwpolyline2.dxf",
    "lwpolyline-closed.dxf",
    "null-polygons.dxf",
    "open-polyline.dxf",
    "polygon8.dxf",
    "polygon-concave.dxf",
    "polygon-concave-hole.dxf",
    "polygon-concave-simple.dxf",
    "polygon-holes-touch.dxf",
    "polygon-intersect.dxf",
    "polygon-many-holes.dxf",
    "polygon-riser.dxf",
    "polygons.dxf"] #THESE ALL SEEM TO WORK

for fn in filenames1:
  print(fn)
  f, c = importEZDXFface(filepath1+fn, flattenlayers=True) #returns faces list and compound
  Part.show(c)`

This is the list of files I got from the forum:

`filenames2=["viga.dxf", #WORKS
    "rectangle_R12.dxf", #WORKS
    #"nintendo-classic-v4.dxf", #WORKS but takes awhile
    #"ncr2lc.dxf", #WORKS but takes awhile
    #"KINETIKMSystem_H930N36_K30Systembauprofil_30x60_2D.dxf", #NOT WORKING bc of "REGION"?
    "hfs5_2040_50_view1_12.dxf", #WORKS but ATTDEF doesn't
    "freecad_to_pad.dxf", #WORKS if flattenlayers=True
    #"FreeCadTestimport_ASCII.dxf", #NOT WORKING? bc of "TEXT", "DIMENSION"?
    #"FreeCadTestimport.dxf", #NOT WORKING? bc of "TEXT", "DIMENSION"?
    #"ETHZ2-2G-O62-01_BOARD-TOP_SIDE.dxf", #NOT WORKING bc of "INSERT"??
    #"ETHZ2-2G-O62-01_BOARD-Bottom_side.dxf",  #NOT WORKING bc of "INSERT"??
    "dxf_import_challenge.dxf", #WORKS
    #"donkey_kong_cab_-_gaetan.dxf", #NOT WORKING? bc of "TEXT", "DIMENSION"?
    "Corel-test-1.dxf", #SEEMS TO WORK
    "611_plate_V1.DXF"] #WORKS`

These are DXFs from the OpenSCAD project that I had trouble with:

`filenames3=["ellipse-arc-rot.dxf", #NOT WORKING
    ##"multiple-layers.dxf", #NOT WORKING BC OF SPECIAL CHAR IN LAYER NAME
    #"polygon-mesh.dxf",#WORKING? FLATTENED???
    "polygon-overlap.dxf",#WORKING??
    "polygon-self-intersect.dxf",#NOT WORKING??
    #"transform-insert.dxf",#NOT WORKING - USES INSERT
    "triangle-with-duplicate-vertex.dxf"] #WORKING?`

mconsidine

KeithSloan, 2023-04-15:
Okay just trying to add you code.

in importCSG.py processDXF needs to create a 2D object and current code is
````
face = importEZDXFface(filename,layer,doc)
    print(f"face {face}")
    obj=doc.addObject('Part::Feature','dxf_%s_%s' % (shortname,layer or "all"))
    obj.Shape=face
````
Now the latest version of importEZDXFface returns a tuple faces, compound or None.

The code has
````
if len(faces)>0:
            #    for f in faces:
            #        Part.show(f)
                compound = Part.makeCompound(faces)
            ##    Part.show(compound)

                return faces, compound # TODO: return layer name, color, labels?
                #return faces[0]
            else:
                print("No wires created")
                return None
````
This does not seem to be valid if it has a single face, does it not return None and the No wires created message.

Might it be better to rename the function importEZDXFshape
and return either faces[0] or compound.Shape

mconsidine, 2023-04-15:
Yes, you are correct.  At first, in order to maintain backward compatitiblity, I had it return just faces[0].

But I wanted to leave open the potential for more to be returned, as I noticed for example on an example DXF that had a hole the "hole" part wasn't an element or discernible by EZDXF.  But, if all the different elements were returned as faces and available to FC then the user could "carry on" by doing whatever extrusions they wanted, boolean cuts, unions, etc etc.

It seemed to me that the 'spirit' of the import command would return all the pieces for the user.  But I understood that that was not the way it was being (or could be) used.

Bottom line - I'll go with whatever you would suggest to both preserve/improve existing functionality while leaving room for the return of something more complex than a face.

On a different note,  I think I was getting myself too tangled up in list-bracket hell so I'm trying to detangle that.  And I figured out how to get at the innards of the item returned by the 'INSERT' entity, but I need to clean up my bracket-hell first...

KeithSloan, 2023-04-15:
Okay, when the importer parses an import function ( or any other function ) it has to create an Object that gets put on the parsing stack. In the case of DXF it is going to be 2D. Now I assume that as the DXF file can have several layers, these get created as a list of faces, so returning a single face ( which is a Part.Shape ) or the Shape of a Compound is goodness.
(I think the old code only worked for a single layer?).

Think I will rename the function as returning a Shape rather than a Face. i.e it could return a single Face which is a Shape or the Shape of a compound. Let's see if that works

KeithSloan, 2023-04-15:
Okay I just updated branch mconsidine

Renamed function to importEZDXFshape ( which either returns a Face or a Compound, both of which are Part Shapes.

Added testDxf files and also made your test script into a Macro - see Macros directory.

If we can do some testing then I will merge with master branch.

Not sure future support of colour might work as from memory color is a ViewObject attribute.

Thanks for all your work todate.

KeithSloan, 2023-04-19:
Now merged into master branch

mconsidine, 2023-04-19:
Thanks.  Just in time for me to be ready to push an update that seems to
deal with INSERT correctly :)

Sent from allegedly smart phone

mconsidine, 2023-04-22:
I've posted an update to my code, which should (finally) correctly deal with rotated ellipse arcs.  It also seems to handle well the INSERT tag, based on my tests.Also uploaded is a zip file with all the DXFs I ran this against.  They were gathered from openscad source files, forum and other web searches.Generally it seems then when there's a fail, it's due to a tag intentionally not being handled, like TEXT or DIMENSION.  But my goal was to try to assist in getting geometry over, not labels, etc.Also, I tried to anticipate the potential for returning a list of lists, e.g. faces organized by layers.  Not how successfully that is done, but the pieces should be there if downstream uses evolve.As it stands, I believe this handles pretty well all the examples provided with openscad.  (There's an example of two intersecting polygons that may still need work, fwiw).Hope this helps,Matt

KeithSloan, 2023-04-22:
Not seeing any code, Pull Request, or change to your Fork, where are you posting the updates.

mconsidine, 2023-04-22:
Ok, I think I just correctly generated the request.  Changes are in the
mconsidine version of my fork ...
Matt

mconsidine, 2023-04-22:
Not sure I did this the right way, but added an additional edit a moment
ago.  So the correct version to check/merge/evaluate is the one labeling as
fixing a typo/edge case in the ellipse routine.  (It turns out if you have
the start point real, real close to the end point of the arc, even though
it supposed to be closed, FC gives you a face that is a bounding box and
not an ellipse ...)
Matt

closed 2023-05-23 by KeithSloan