KeithSloan/OpenSCAD_Alt_Import #14: Import DXF and DXFObjects @mconsidine

KeithSloan, 2023-04-24:
Matt this branch DxfObjects is an experiment

1) On importing/Opening a DXF file it should off extra options on the import.

New Importer : DXF (importDXF)

Will invoke importDXF.py

Which just calls your loadDXF function, what is needed is to create or add the DXF file to current document.

I guess we could just create a Shape with importEZDXFshape but wonder if more that one layer/face should
not create a compound but leave as separate shapes. You have more experience of what people might want.

i.e. This function just needs to create a FC document from the DXF file.

2) Another extra option is

New Importer : DXFObject (importDXFObj)

This will invoke importDXFObj which will create a DXFObject in a similar way to a SCADObject

An external DXFeditor can be configured by

````
FreeCAD | Tools | Edit parameters

  Group : Base App
      Preferences
         Mod
            OpenSCAD
                externalDXFeditor
                    Text : <path to editor > 
                    
On my Mac it creates the DXFObject okay but when I try to edit the object by setting the objects edit property to True,
it does not work at least with Inkscape

I suspect that when I invoke Inkscape I need to invoke it an inkscape document rather than a dxf file.

Thoughts

mconsidine, 2023-04-24:
Just pulled and checked this branch.  Line 292 of OpenSCADdxf.py  should start
  def importEZDXFface
(as opposed to importEZDXFshape)
I ran (almost) all the DXF examples from the OpenSCAD source folder through the Add element interface and did not get any unexpected fails.  (There are a couple of those files which use TEXT, DIMENSION, etc, which basically ignored)

I have not been using an external editor, but will try to configure that setup.  But, yes, I think Inkscape wants to see its own format or SVG, perhaps.

I think there's more that can be done to clean up the main routine.  I'm not convinced all the "flattening" needs to get done.  Candidly, I'm agnostic about whether a compound comes back or a compound plus a list of faces.  It occurred to me that the latter could be useful if one extruded all the faces and then was able to "Cut" holes or whatever.  Then again, I suppose if someone is wanting to pull in a DXF they just want the geometry and lines in order to recreate the layout.  And for that a compound would be fine (??? dunno).

In any case, I'm sure there will be cases people come up with that make the importing a bit misbehaved, given that the DXF format seems to be a bit of a mess.  But maybe this moves the ball down the field a bit ...

Thanks for all you help!

Matt

KeithSloan, 2023-04-24:
Actually I think it should be called importEZDXFshape as it may return a Compound which is a Shape not a Face.

But all code should refer to the same name and currently it does not.

mconsidine, 2023-04-24:
Whatever you would like it called is fine by me!

KeithSloan, 2023-04-24:
Just tried with external editor = LibreCAD and it seems to work.

Think with Inkscape there is a clash with Python and FC.

mconsidine, 2023-04-24:
Still catching up ... I'll try getting the external set up on Wed (I teach Tues/Thurs, which shoots those days).  I'm running on Linux so we'll see if the same kind of clash shows up.  Also, I'm running self-compiled vs an Appimage ...

KeithSloan, 2023-04-25:
What would be good is if you can update importDXF.py so that it uses your code to create or import to a document.

Also wondering if we should look at supporting when importing as a DXFobject that there is the option to create each layer as a separate DXFobject rather than one Shape.

KeithSloan, 2023-04-25:
Not seeing the wood for the trees.

How about as a way forward
importEZDXFshape gets renamed importEZDXFfaces and returns a list of tuples, layerName, face.

New function importEZDXFshape that calls importEZDXFfaces and if there is only one face returns it or makes a compound and returns it.

Then when dealing with OpenSCAD dxf files just carry on using importEZDXFshape

But for importDXF can call importDXFfaces and then run through the list of tuples adding to the document.

For creating DXFObjects can again call importDXFfaces and the run though the list of tuples creating a DXFObject for each Face.

What do you think?

mconsidine, 2023-04-25:
I think that makes sense.  I should have time tomorrow to check that and importdxf.pyI used a "clean" approach to a come up with a version that just took everything on all layers and returned that.  It actually worked better on a couple of DXFs and now I have to figure out why, as it also failed on an important one ... And I have to figure out why.Another wrinkle is what to do with POINTS.  I've tried collecting them in a separate structure and then turning them into polygons when a different type of entity is encountered. That might have promise in at least preserving that info.  But it also means that non-geometric info comes back, like arrow heads from dimension lines.

mconsidine, 2023-05-03:
FYI, the DXF that is causing a problem here  https://forum.freecad.org/viewtopic.php?t=78003loads fine with my new code.I don't believe that anything in the main routine returns anything different.  So I can clean it all up a bit more and submit a pull request if you are up for the merge.Rgds,Matt

mconsidine, 2023-05-03:
This should have been attached.

KeithSloan, 2023-05-04:
Should be good for PR requests after 10th May.

mconsidine, 2023-05-10:
PR submitted.  Comments there show tests that I did.