KeithSloan/OpenSCAD_Alt_Import #19: ScadFileObject and ScadObject imports not working

akrly, 2023-07-20:
I had a the translation issue happening to me when I imported .scad files into a a basic FreeCAD installation. So I tried and install this alternative import tool. The translation issue disappeared, but only the New Importer: SCAD (importCSG) works. I get the following error when I try to use the ScadFileObject or ScadObject options:
```
09:28:31  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'AttributeError'>: module 'importFileSCAD' has no attribute 'insert'
```

I also tried to import some of the basic designs, such as the openscad logo, but  I get the same error.

KeithSloan, 2023-07-20:
I am getting a different error when I try and import logo.scad.

Have created a branch called fix, are you able to test? or do I need to merge with master.

KeithSloan, 2023-07-20:
Okay have merged fix, give that a try, like In said I get a different error so maybe have not fixed your problem.

KeithSloan, 2023-07-26:
Any chance you could upload the file or email ( address should be in README)

akrly, 2023-07-26:
I tried it with the[ logo.scad](https://files.openscad.org/examples/Basics/logo.html) file.

Also tried it with one of my designs. It's more complex, but you can find it [here](https://github.com/akrly/cnc-wasteboard). You only need the scad files in the root of the repo. The main.scad generates the body, but it uses all other scad files.

KeithSloan, 2023-07-26:
Not seeing the problem here on ScadFileObject

Screen shot when importing as a

New Importer : ScadFileObject (importFIleSCAD ) 
[image]

My FreeCAD info is 

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: C/Default (C)
Installed mods: 
  * ImportNURBS 1.1.0
  * ImportExport_3DM 0.1.0
  * DynamicData 2.46.0
  * Pyramids-and-Polyhedrons
  * occi-freecad-plugin 0.1.0
  * GDML 2.0.0
  * Design456 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1
  * Curves 0.6.8
[/code]

With New Importer SCADObject (importSCAD) I get problems.

This is because the way it works for SCADObjects is to copy the source scad file into FreeCAD but does not copy all the imported files. I need to work out a way of handling this see issue #20.

I have had problems with updates via the Addon Manager in the past, it is important to clear the Addon Manager cache.

akrly, 2023-07-27:
Thanks for the tip about the sticky cache. I'll try it with a clean install, just to be sure and report back.

akrly, 2023-07-27:
I tried all three new imports.

New Importer : SCAD (importCSG)
[image]

New Importer : ScadFileObject (importFileSCAD)
[image]

New Importer : (importSCAD)
[image]

My FreeCAD info:

[code]
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Alternate_OpenSCAD 1.0.0
[/code]

KeithSloan, 2023-07-27:
Not sure why this should be happening and I cannot reproduce. Have asked in the FreeCAD forum [https://forum.freecad.org/viewtopic.php?t=79986](url)

KeithSloan, 2023-07-28:
Please could you open the file importFileSCAD.py or importSCAD.py and check the start of the file, should look the same as
[https://github.com/KeithSloan/OpenSCAD_Alt_Import/blob/master/importFileSCAD.py](url)
or [https://github.com/KeithSloan/OpenSCAD_Alt_Import/blob/master/importSCAD.py](url)

akrly, 2023-07-29:
This is the comparison of the importFileSCAD. The left is from my local directory, the right one is from the link above. The only difference is a new line at the bottom.

[image]

KeithSloan, 2023-08-02:
Don't see why it should be any different but FreeCAD 0.21 has just been released just wondering if it makes any difference on Ubuntu 23.04. On the Mac its working just the same see.

For FreeCAD 0.21 see [https://github.com/FreeCAD/FreeCAD/releases](url)

mconsidine, 2023-08-03:
I believe I have a fix/workaround for this in the pull request I provided from my DxfObjects branch.
mconsidine

KeithSloan, 2023-08-04:
Thanks Matt seems to be working for me but have an issue with the No Parking DXF file see new issue #22.

KeithSloan, 2023-08-04:
@akrly Interested to know if the latest version works for you on your OS.

Also is it okay to share your cnc-wasteboard as this shows up the problem of import of files with use and include

akrly, 2023-08-04:
> @akrly Interested to know if the latest version works for you on your OS.
> 
> Also is it okay to share your cnc-wasteboard as this shows up the problem of import of files with use and include issue #20

I'll test it tonight and sure feel free to use the design. This reminded me that I should include some sort of a licence, so it's more obvious :)

([link](https://github.com/akrly/cnc-wasteboard))

mconsidine, 2023-08-04:
When I try to process main.scad from the command line, I get an error (ie outside of FreeCAD) :+1: 
`matt@matt-Precision-Tower-7910:~$ /home/matt/Downloads/openscad-manifold/build/openscad -o /tmp/matt1.csg /home/matt/Downloads/DXFtests/scadtests/main.scad
ERROR: Parser error: syntax error in file Downloads/DXFtests/scadtests/main.scad, line 88
Can't parse file '/home/matt/Downloads/DXFtests/scadtests/main.scad'!

It looks like the problem may be with OpenSCAD processing?

`

mconsidine, 2023-08-04:
Line 88 of main.scad has a typo.  There are two "," in the argument next to each other and there should only be one.

KeithSloan, 2023-08-04:
The way scadObject works is that it copies the file to a directory were FC stores other data related to the document.

What I need to do is add code to look at the main scad file and where there are use/include statements try and copy those files as well. Then when the main scad file gets passed to OpenSCAD via the subprocess, also add the directory to the external parameter OpenSCADPath and pass that to the openscad call.

mconsidine, 2023-08-04:
Understood.  Conceivably, though, that would involve copying full OpenSCAD libraries, no?  E.g. I have a file that has at the start:
`use <threadlib/threadlib.scad>
use <threads.scad>
//use <ruler.scad>
include <Chamfers-for-OpenSCAD/Chamfer.scad>;
`
Could include/use files be opened from their original location in a read-only mode?  Is the reason for copying the source .scad files to a tmp directory to avoid any file corruption?

akrly, 2023-08-05:
@KeithSloan 

I cloned the OpenSCAD_Alt_Import repo and used the  [FreeCAD_0.21.0-Linux-x86_64.AppImage](https://github.com/FreeCAD/FreeCAD/releases/download/0.21.0/FreeCAD_0.21.0-Linux-x86_64.AppImage). The importFileSCAD works, but the importSCAD doesn't. It just renames the the file and stops.

**This is importFileSCAD**

[image]

**and this is using importSCAD**

[image]

KeithSloan, 2023-08-05:
Good that importFileSCAD now works.

The current problem with importSCAD is when the file has use or includes see issue #20, which requires additional code to handle these functions.