KeithSloan/OpenSCAD_Alt_Import #29: Proposal - Use Linked objects when the original file uses `children()`

EmperorArthur, 2024-08-30:
# Rationale

Working on https://github.com/kennetek/gridfinity-rebuilt-openscad/ one of the common things we do is create a linear pattern of objects.  Each of these objects is identical, but they are translated.

From a parametric standpoint, it would make things easier to replicate this structure.

# Challenges
I recognize this may not be possible without additional annotations and/or scans to detect if two items in the intermediate csg file are identical.  It would be nice though...

KeithSloan, 2024-08-30:
Thanks for your proposal.

FreeCAD OpenSCAD uses an importer for csg files which it parses. 

When OpenSCAD processes the scad source to produce the csg, information on children has been lost.

If I look at this example

`
//Use only the first child, multiple times
  
module lineup(num, space) {
   for (i = [0 : num-1])
     translate([ space*i, 0, 0 ]) children(0);
}

lineup(5, 65){ sphere(30);cube(35);}

`
It produces in OpenSCAD

[image]

If I look at the exported csg file it looks like
```
group() {
	group() {
		multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			group() {
				sphere($fn = 0, $fa = 12, $fs = 2, r = 30);
			}
		}
		multmatrix([[1, 0, 0, 65], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			group() {
				sphere($fn = 0, $fa = 12, $fs = 2, r = 30);
			}
		}
		multmatrix([[1, 0, 0, 130], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			group() {
				sphere($fn = 0, $fa = 12, $fs = 2, r = 30);
			}
		}
		multmatrix([[1, 0, 0, 195], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			group() {
				sphere($fn = 0, $fa = 12, $fs = 2, r = 30);
			}
		}
		multmatrix([[1, 0, 0, 260], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			group() {
				sphere($fn = 0, $fa = 12, $fs = 2, r = 30);
			}
		}
	}
}
```

It would be helpful if OpenSCAD supported adding comments to the CSG, which the importer could parse as additional csg statements, maybe you could lobby OpenSCAD for such a feature.

But having said that there might be the possibilty to scan the csg file for multimatrix that has the same function code,.
I will have a think about that, certainly not straight forward with current simple lex parser.

KeithSloan, 2024-08-31:
Are you able to test a new branch **children** ?

[testChildren2.csg.txt](https://github.com/user-attachments/files/16825589/testChildren2.csg.txt)
[testChildern2.FCStd.txt](https://github.com/user-attachments/files/16825590/testChildern2.FCStd.txt)

Thanks once again for proposal

EmperorArthur, 2024-09-01:
> Are you able to test a new branch **children** ?

Need to find where FreeCAD puts these files on Windows, then sure.

I will update once I test this.

KeithSloan, 2024-09-01:
https://wiki.freecad.org/Installing_more_workbenches

says On Windows it is usually C:\Program Files\FreeCAD\Mod\

EmperorArthur, 2024-09-01:
That's the directory for things which come bundled with the FreeCAD installer.  Normal Users don't have write permissions to the "C:\Program Files\" directory on windows.  So, FreeCAD mod manager puts them in "C:\Users\<username>\AppData\Roaming\FreeCAD\Mod\"

In this case it's "C:\Users\<username>\AppData\Roaming\FreeCAD\Mod\Alternate_OpenSCAD\".  Found using "Help" -> "Automatic Python Modules Documentation"

Attempting the changes now.

EmperorArthur, 2024-09-01:
# children Branch
[image]

Error:
```
14:04:49  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\EmperorArthur\AppData\Roaming\FreeCAD\Mod\Alternate_OpenSCAD\.\importAltCSG.py", line 106, in open
    processCSG(doc, tmpfile)
  File "C:\Users\EmperorArthur\AppData\Roaming\FreeCAD\Mod\Alternate_OpenSCAD\.\importAltCSG.py", line 181, in processCSG
    result = parser.parse(f.read())
  File "C:\Program Files\FreeCAD 0.21\bin\lib\site-packages\ply\yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\site-packages\ply\yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "C:\Users\EmperorArthur\AppData\Roaming\FreeCAD\Mod\Alternate_OpenSCAD\.\importAltCSG.py", line 221, in p_group_action1
    p[0] = [fuse(p[5],"Group")]
  File "C:\Users\EmperorArthur\AppData\Roaming\FreeCAD\Mod\Alternate_OpenSCAD\.\importAltCSG.py", line 676, in fuse
    myfuse.Shape = myfuse.Base.Shape.fuse(myfuse.Tool.Shape)
<class 'AttributeError'>: 'App.DocumentObject' object has no attribute 'Shape'
14:05:00  Recompute failed!
```

# master Branch (Working)
[image]

```
14:11:44  Verbose = False
14:11:44  Processing : D:/src/gridfinity-rebuilt-openscad/gridfinity-rebuilt-bins.scad
14:11:44  Calling OpenSCAD
14:11:44  
ECHO: height_mm = 37.25

14:11:44  Using Alternate OpenSCAD Importer
14:11:44  Doc gridfinity_rebuilt_bins useMaxFn 16
14:11:44  obj [] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  performMultMatrix LinearExtrude
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [] has no Name & Shape
14:11:44  performMultMatrix square001
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  performMultMatrix square002
14:11:44  union
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  performMultMatrix union
14:11:44  performMultMatrix square003
14:11:44  performMultMatrix union
14:11:44  performMultMatrix square003
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  performMultMatrix LinearExtrude001
14:11:44  performMultMatrix square004
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  performMultMatrix square005
14:11:44  union
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  performMultMatrix union001
14:11:44  performMultMatrix square006
14:11:44  performMultMatrix union001
14:11:44  performMultMatrix square006
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
[... 691 lines cut ...]
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  union
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  performMultMatrix union036
14:11:44  performMultMatrix union036
14:11:44  obj [] has no Name & Shape
14:11:44  performMultMatrix cube046
14:11:44  performMultMatrix cube047
14:11:44  performMultMatrix cylinder047
14:11:44  union
14:11:44  performMultMatrix union037
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  union
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  performMultMatrix union037
14:11:44  performMultMatrix union037
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  performMultMatrix difference013
14:11:44  performMultMatrix Group022
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  obj [<Part::PartFeature>] has no Name & Shape
14:11:44  End processing CSG file
```

KeithSloan, 2024-09-01:
Please could you attach the csg file ( You will have to add something like .txt to its name.

EmperorArthur, 2024-09-01:
[gridfinity-rebuilt-bins.csg.txt](https://github.com/user-attachments/files/16830154/gridfinity-rebuilt-bins.csg.txt)

Default bin configuration from: https://github.com/kennetek/gridfinity-rebuilt-openscad

KeithSloan, 2024-09-01:
Thanks I was hoping that having stripped the objects name of any last three numerals, if the name was the same as one previously created and reference stored in a dictionary, the objects were the same apart from different Placement.

KeithSloan, 2024-09-01:
Another complication is that things like fuse works on objects with Shape but if the Object is a Linked object which has no   Shape, have to use LinkedObject, need to change quite a bit of code, may take a while.

KeithSloan, 2024-09-01:
I think we have a problem Houston. 

If you look at your problem report it fails at
File "C:\Users\EmperorArthur\AppData\Roaming\FreeCAD\Mod\Alternate_OpenSCAD\.\importAltCSG.py", line 676, in fuse
    myfuse.Shape = myfuse.Base.Shape.fuse(myfuse.Tool.Shape)
<class 'AttributeError'>: 'App.DocumentObject' object has no attribute 'Shape'
14:05:00  Recompute failed!

The object with no attribute  'Shape' is a LinkedObject.

I don't think that it is the sort of thing a user might think of doing, so I doubt that it has ever been raised as a bug but by trying to treat objects which are just the same but with different Placement as a linked Objects, we have possibly hit a problem. Let's say the operation is a fuse, i.e. a union, what the parsed code is trying to do is a fuse, where one of the objects is a linked object. One can't even adopt an approach of well its a linkedObject, so use the Shape of the linked object in the fuse, as its Shape is not in the right place. One would have to create a Shape from the info in the linked object and its Placement in any boolean operation, which a makes things complicated and is it worth it?
If the user starts to change what is in the FreeCAD file that might well break things, the fuse will be using a made up Shape rather than an original Shape. 

Don't suppose you feel like creating a test case in native FC, which involves a Linked Object and a Boolean operation.
I think it will most likely throw up a bug.

Obviously if FC was able to handle booleans where objects might be linked object, I think we have a solution but I strongly suspect this is not the case currently

EmperorArthur, 2024-09-01:
Sounds like I need to dig into the internals a bit more.

The other thing is I'm running the latest stable, and FreeCAD 1.0 RC1 is next week.  That includes the toponaming work, which is a significant amount of internal changes.  The bug might be fixed there, but this importer may need updates to support it.

closed 2025-03-18 by KeithSloan