KeithSloan/OpenSCAD_Alt_Import #31: "No module named pkg_resources." Undeclared dependency package?

BurntVoxel, 2024-12-13:
Hi, I wanted to try out your addon but I can't get it started for some reason. 

I *am* using NixOS, so my setup may be a little weird, but I did figure out how to install the dependencies the Addon Manager called for. It didn't let me install it until I had set up the library `ezdxf`, for one. 

Yet, even with that, the extension crashes on startup. 
```
12:27:02  During initialization the error "No module named 'pkg_resources'" occurred in freecad.OpenSCAD_Alt_Import
12:27:02  --------------------------------------------------------------------------------
12:27:02  Traceback (most recent call last):
  File "<string>", line 232, in InitApplications
  File "/nix/store/nmqxyr00in2arwrq5qd1qipsanz1yrn5-python3-3.11.10/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/nomad/.local/share/FreeCAD/Mod/Alternate_OpenSCAD/./freecad/OpenSCAD_Alt_Import/__init__.py", line 2, in <module>
    __import__("pkg_resources").declare_namespace(__name__)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/vv9rp0yfw8b8jc9ssg7mgn9xrgrfsyai-shiboken2-5.15.15/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'pkg_resources'
12:27:02  --------------------------------------------------------------------------------
```
```
OS: NixOS 24.11 (Vicuna) (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.0.0.Unknown
Build type: Release
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.2.6, OCC 7.6.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FreeCAD-Ribbon 1.4.16
  * Alternate_OpenSCAD 1.0.0
  * SearchBar 1.2.1
  * cadquery_module 2.0.0
  * Render 2024.11.22
```
```
> pip list  
Package           Version
----------------- ------------
ezdxf             1.3.2
fonttools         4.54.1
lxml              5.3.0
numpy             1.26.4
pip               24.0
pyparsing         3.1.4
setuptools        75.1.1.post0
typing_extensions 4.12.2
```
I'm not actually sure what other info would help. Ask anything.

KeithSloan, 2024-12-13:
> Hi, I wanted to try out your addon but I can't get it started for some reason.
> 
> I _am_ using NixOS, so my setup may be a little weird, but I did figure out how to install the dependencies the Addon Manager called for. It didn't let me install the addon until I had set up the library `ezdxf`, for one.
> 
> Yet, even with that, the extension crashes on startup.
> 
> ```
> 12:27:02  During initialization the error "No module named 'pkg_resources'" occurred in freecad.OpenSCAD_Alt_Import
> 12:27:02  --------------------------------------------------------------------------------
> 12:27:02  Traceback (most recent call last):
>   File "<string>", line 232, in InitApplications
>   File "/nix/store/nmqxyr00in2arwrq5qd1qipsanz1yrn5-python3-3.11.10/lib/python3.11/importlib/__init__.py", line 126, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
>   File "<frozen importlib._bootstrap_external>", line 940, in exec_module
>   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
>   File "/home/nomad/.local/share/FreeCAD/Mod/Alternate_OpenSCAD/./freecad/OpenSCAD_Alt_Import/__init__.py", line 2, in <module>
>     __import__("pkg_resources").declare_namespace(__name__)
>     ^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/nix/store/vv9rp0yfw8b8jc9ssg7mgn9xrgrfsyai-shiboken2-5.15.15/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> ModuleNotFoundError: No module named 'pkg_resources'
> 12:27:02  --------------------------------------------------------------------------------
> ```
> 
>  ```
> OS: NixOS 24.11 (Vicuna) (KDE/plasmax11/xcb)
> Architecture: x86_64
> Version: 1.0.0.Unknown
> Build type: Release
> Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.2.6, OCC 7.6.2
> Locale: English/United States (en_US)
> Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
> Installed mods: 
>   * FreeCAD-Ribbon 1.4.16
>   * Alternate_OpenSCAD 1.0.0
>   * SearchBar 1.2.1
>   * cadquery_module 2.0.0
>   * Render 2024.11.22
> ```
> 
>  ```
> > pip list  
> Package           Version
> ----------------- ------------
> ezdxf             1.3.2
> fonttools         4.54.1
> lxml              5.3.0
> numpy             1.26.4
> pip               24.0
> pyparsing         3.1.4
> setuptools        75.1.1.post0
> typing_extensions 4.12.2
> ```
> 
> I'm not actually sure what other info would help. Ask anything.

The packages, python libraries that it uses are, see metadata file
```
workbenches=OpenSCAD
pylibs=ply,ezdxf
```
So ply and ezdxf

The message 
```
File "/nix/store/vv9rp0yfw8b8jc9ssg7mgn9xrgrfsyai-shiboken2-5.15.15/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> ModuleNotFoundError: No module named 'pkg_resources'
```
When it says ModuleNotFound it could be looking for ply given it looks like you have ezdxf installed.

But if it is an issue with the addon manager and nix, I would try

From the freecad console
```
import sys
print(sys.path)
```
Then choose the most appropriate path

On my MacOS I would use
```
/Applications/FreeCAD 1.0.0.app/Contents/Resources/lib/python3.11/site-packages
```

Now FreeCAD is using python3.11 so I make sure in command line mode I am using python3.11
so on a Mac using howbrew
```
brew install python@3.11
```
Then 
```
python3.11 -m pip install ply -t '/Applications/FreeCAD 1.0.0.app/Contents/Resources/lib/python3.11/site-packages'
```
Let me know how you get on.

BurntVoxel, 2024-12-14:
I tried again after simply installing `ply`, no change. Hm. 

Printing sys.path... I'll format it for you, because there's a quirk on Nix that makes this string really long: most packages are packed into a folder with some hash for a prefix. Example, one entry looks like `/nix/store/1gndczpr8inmjn6b83xv7dryszmfjspi-freecad-1.0.0/Mod/Web`. 

Question for you, some of these path entries are duplicates, is that weird? 
```
# FreeCAD
# >>> print(sys.path)

/home/nomad/.local/share/FreeCAD/{
    ./AdditionalPythonPackages, 
    ./AdditionalPythonPackages/py311, 
    ./Macro, # x2 
    ./Mod/Alternate_OpenSCAD, # x2
    ./Mod/cadquery_module, # x2 
    ./Mod/FreeCAD-Ribbon, # x2
    ./Mod/FreeCAD-Ribbon/.Backups, # x4
    ./Mod/FreeCAD-Ribbon/./Resources/icons, # x5
    ./Mod/FreeCAD-Ribbon/./Resources/packages
    ./Mod/FreeCAD-Ribbon/./Resources/stylesheets, # x5
    ./Mod/FreeCAD-Ribbon/./Resources/ui, # x5
    ./Mod/Render, # x2
    ./Mod/SearchBar, # x2 
}, 

/nix/store/{
    ./*-freecad-1.0.0/Ext, 
    ./*-freecad-1.0.0/lib, 
    ./*-freecad-1.0.0/Macro, 
    ./*-freecad-1.0.0/Mod, 
    ./*-freecad-1.0.0/Mod/AddonManager, 
    ./*-freecad-1.0.0/Mod/Assembly, 
    ./*-freecad-1.0.0/Mod/BIM, 
    ./*-freecad-1.0.0/Mod/CAM, 
    ./*-freecad-1.0.0/Mod/Draft, 
    ./*-freecad-1.0.0/Mod/Drawing, 
    ./*-freecad-1.0.0/Mod/Fem, 
    ./*-freecad-1.0.0/Mod/Help, 
    ./*-freecad-1.0.0/Mod/Idf, 
    ./*-freecad-1.0.0/Mod/Import, 
    ./*-freecad-1.0.0/Mod/Inspection, 
    ./*-freecad-1.0.0/Mod/Material, 
    ./*-freecad-1.0.0/Mod/Measure, 
    ./*-freecad-1.0.0/Mod/Mesh, 
    ./*-freecad-1.0.0/Mod/MeshPart, 
    ./*-freecad-1.0.0/Mod/OpenSCAD, 
    ./*-freecad-1.0.0/Mod/Part, 
    ./*-freecad-1.0.0/Mod/PartDesign, 
    ./*-freecad-1.0.0/Mod/Plot, 
    ./*-freecad-1.0.0/Mod/Points, 
    ./*-freecad-1.0.0/Mod/ReverseEngineering, 
    ./*-freecad-1.0.0/Mod/Robot, 
    ./*-freecad-1.0.0/Mod/Show, 
    ./*-freecad-1.0.0/Mod/Sketcher, 
    ./*-freecad-1.0.0/Mod/Spreadsheet, 
    ./*-freecad-1.0.0/Mod/Start, 
    ./*-freecad-1.0.0/Mod/Surface, 
    ./*-freecad-1.0.0/Mod/TechDraw, 
    ./*-freecad-1.0.0/Mod/Test, 
    ./*-freecad-1.0.0/Mod/Tux, 
    ./*-freecad-1.0.0/Mod/Web, 
    ./*-pyside2-5.15.15/lib/python3.11/site-packages, 
    ./*-python3.11-contourpy-1.3.0/lib/python3.11/site-packages, 
    ./*-python3.11-cycler-0.12.1/lib/python3.11/site-packages, 
    ./*-python3.11-ddt-1.7.2/lib/python3.11/site-packages, 
    ./*-python3.11-fonttools-4.54.1/lib/python3.11/site-packages, 
    ./*-python3.11-gitdb-4.0.11/lib/python3.11/site-packages, 
    ./*-python3.11-gitpython-3.1.43/lib/python3.11/site-packages, 
    ./*-python3.11-kiwisolver-1.4.7/lib/python3.11/site-packages, 
    ./*-python3.11-matplotlib-3.9.2/lib/python3.11/site-packages, 
    ./*-python3.11-numpy-1.26.4/lib/python3.11/site-packages, 
    ./*-python3.11-packaging-24.1/lib/python3.11/site-packages, 
    ./*-python3.11-pillow-11.0.0/lib/python3.11/site-packages, 
    ./*-python3.11-pivy-0.6.9/lib/python3.11/site-packages, 
    ./*-python3.11-ply-3.11/lib/python3.11/site-packages, 
    ./*-python3.11-pybind11-2.13.6/lib/python3.11/site-packages, 
    ./*-python3.11-pycollada-0.8/lib/python3.11/site-packages, 
    ./*-python3.11-pyparsing-3.1.4/lib/python3.11/site-packages, 
    ./*-python3.11-python-dateutil-2.9.0.post0/lib/python3.11/site-packages, 
    ./*-python3.11-pyyaml-6.0.2/lib/python3.11/site-packages, 
    ./*-python3.11-scipy-1.14.1/lib/python3.11/site-packages, 
    ./*-python3.11-six-1.16.0/lib/python3.11/site-packages, 
    ./*-python3.11-smmap-6.0.0/lib/python3.11/site-packages, 
    ./*-python3.11-tkinter-3.11.10/lib/python3.11/site-packages, 
    ./*-python3-3.11.10/lib/python3.11, 
    ./*-python3-3.11.10/lib/python3.11/lib-dynload, 
    ./*-python3-3.11.10/lib/python3.11/site-packages, 
    ./*-python3-3.11.10/lib/python311.zip, 
    ./*-shiboken2-5.15.15/lib/python3.11/site-packages, 
}
```

KeithSloan, 2024-12-14:
I think you need to ask in the FreeCAD forum https://forum.freecad.org

I asked ChatGPT and it came up with the following ( Note ChatGPT does not always know the correct answers)

[image]
[image]
[image]
[image]